mvcisback/py-aiger #34: Handling of latches is unreasonably inefficient

MarkusRabe, 2018-07-28:
aig.cutlatches() and aig.latches.items() appear to take forever for a file with around 15k gates and a couple of hundred latches. Link to file: [https://drive.google.com/open?id=18tV9r7Fqf99qFDcn5rTSfrDYZHXJ3WW7](url)

mvcisback, 2018-07-28:
@MarkusRabe yeah, this is certainly a pain point of the current design. I need to traverse through the circuit to the latches and change them to inputs. This could be made slightly better using a hand optimized traversal compared to lens. That said I think a better solution is the following idea I've been playing with:

1. Remove the type of LatchIn. LatchIn and Inputs are only distinguished by a lookup table saying whether that input is or is not a latch.
2. Cut latches is now just a manipulation of said lookup table.

MarkusRabe, 2018-07-28:
Sounds like a good design.

mvcisback, 2018-08-04:
Fixed in #51

closed 2018-08-04 by mvcisback