mvcisback/py-aiger #38: BoolExpr DSL

mvcisback, 2018-07-28:
Writing simple combinatorial is clumsier than it need be. Since feedback and other sequential features are not required, I propose creating a DSL similar to the BV DSL for writing simple combinatorial circuits.

It could look something like:
```python
x, y = BoolExpr('x'), BoolExpr('y')
a = (x & y) | ((~x) | y)
b = x >> y  # x implies y
c = x << y # y implies x
d = x == y # x `xnor` y
e = x ^ y # x `xor` y
```

MarkusRabe, 2018-07-28:
I'd suggest not using >> for implications, but <. This may avoid nasty
confusions when handling BoolExpr and BV at the same time.

mvcisback, 2018-07-29:
@MarkusRabe Yeah, thats a good idea.

mvcisback, 2018-07-30:
Implemented in 3bcd2f8b4589d24b9d21c028a7abf45498a4b611

closed 2018-07-30 by mvcisback