mvcisback/py-aiger #74: Printing aig crashes because it hits recursion limit

MarkusRabe, 2018-08-12:
See stack trace below. Can we make __repr__ non-recursive?

>  File "/Users/markus/work/py-aiger/py-aiger-analysis/aiger_analysis/abc.py", line 12, in simplify
>     extract_aig(e).write(aag_name)
>   File "/Users/markus/work/py-aiger/py-aiger/aiger/aig.py", line 236, in write
>     f.write(repr(self))
>   File "/Users/markus/work/py-aiger/py-aiger/aiger/aig.py", line 79, in __repr__
>     return repr(self._to_aag())
>   File "/Users/markus/work/py-aiger/py-aiger/aiger/parser.py", line 256, in aig2aag
>     AAG({}, {}, {}, [], circ.comments),
>   File "/Users/markus/work/py-aiger/py-aiger/aiger/parser.py", line 294, in _to_aag
>     [c], aag, max_idx=max_idx, lit_map=lit_map
>   File "/Users/markus/work/py-aiger/py-aiger/aiger/parser.py", line 294, in _to_aag
>     [c], aag, max_idx=max_idx, lit_map=lit_map
>   File "/Users/markus/work/py-aiger/py-aiger/aiger/parser.py", line 294, in _to_aag
>     [c], aag, max_idx=max_idx, lit_map=lit_map
>   [Previous line repeated 2981 more times]
>   File "/Users/markus/work/py-aiger/py-aiger/aiger/parser.py", line 290, in _to_aag
>     for c in fn.mapcat(lambda g: g.children, gates):
>   File "/Users/markus/work/py-aiger/py-aiger/aiger/parser.py", line 290, in <lambda>
>     for c in fn.mapcat(lambda g: g.children, gates):
>   File "/Users/markus/work/py-aiger/py-aiger/aiger/aig.py", line 22, in children
>     return (self.left, self.right)
> RecursionError: maximum recursion depth exceeded while calling a Python object

mvcisback, 2018-08-21:
This is surprising. I guess this means the depth of the circuit is very large. I think this should be possible. I've been meaning to rewrite this code (it's a bit of wart atm).

mvcisback, 2020-11-11:
Should be fixed :)

closed 2020-11-11 by mvcisback