mvcisback/py-aiger #77: Problems reading large file

MarkusRabe, 2018-08-21:
Here's a file that aiger seems to have problems to read.

[tmp.aag.txt](https://github.com/mvcisback/py-aiger/files/2304338/tmp.aag.txt

First, it takes a long time, and then it crashes with the error below. Recursion limit reached.

Wrote AIG certificate with 111908 gates to /var/folders/k9/5nkgyr_s7994y8p0fszk6yg80000gn/T/tmpebm6ttu1/result.aag
Traceback (most recent call last):
  File "./safety_game.py", line 94, in <module>
    verbose=args.verbose)
  File "./safety_game.py", line 61, in is_realizable
    verbose=verbose)
  File "/Users/markus/work/py-aiger/py-aiger-analysis/aiger_analysis/cadet.py", line 77, in eliminate
    verbose=verbose)
  File "/Users/markus/work/py-aiger/py-aiger-analysis/aiger_analysis/cadet.py", line 70, in _call_cadet
    verbose=verbose)
  File "/Users/markus/work/py-aiger/py-aiger-analysis/aiger_analysis/cadet.py", line 45, in _call_cadet_on_file
    ret = aa.abc.simplify(ret)
  File "/Users/markus/work/py-aiger/py-aiger-analysis/aiger_analysis/abc.py", line 13, in simplify
    extract_aig(e).write(aag_name)
  File "/Users/markus/work/py-aiger/py-aiger/aiger/aig.py", line 236, in write
    f.write(repr(self))
  File "/Users/markus/work/py-aiger/py-aiger/aiger/aig.py", line 79, in __repr__
    return repr(self._to_aag())
  File "/Users/markus/work/py-aiger/py-aiger/aiger/parser.py", line 256, in aig2aag
    AAG({}, {}, {}, [], circ.comments),
  File "/Users/markus/work/py-aiger/py-aiger/aiger/parser.py", line 294, in _to_aag
    [c], aag, max_idx=max_idx, lit_map=lit_map
  File "/Users/markus/work/py-aiger/py-aiger/aiger/parser.py", line 294, in _to_aag
    [c], aag, max_idx=max_idx, lit_map=lit_map
  File "/Users/markus/work/py-aiger/py-aiger/aiger/parser.py", line 294, in _to_aag
    [c], aag, max_idx=max_idx, lit_map=lit_map
  [Previous line repeated 979 more times]
  File "/Users/markus/work/py-aiger/py-aiger/aiger/parser.py", line 290, in _to_aag
    for c in fn.mapcat(lambda g: g.children, gates):
  File "/Users/markus/work/py-aiger/py-aiger/aiger/parser.py", line 290, in <lambda>
    for c in fn.mapcat(lambda g: g.children, gates):
  File "/Users/markus/work/py-aiger/py-aiger/aiger/aig.py", line 22, in children
    return (self.left, self.right)
RecursionError: maximum recursion depth exceeded while calling a Python object

mvcisback, 2018-08-25:
Seems like the crash is due to #74. Will need to profile reading to determine if slowness is in the PEG parser, the `AAG -> AIG` conversion, or the `AIG -> AAG` conversion for printing.

mvcisback, 2020-11-11:
@MarkusRabe This should be fixed now :)

Finally found some time to re-write the parser/printer to not be recursive and better utilize other machinery we've built up.

closed 2020-11-11 by mvcisback