mvcisback/py-aiger #115: Error loading '.aig' files: 'utf-8' codec can't decode byte

allrtaken, 2020-02-04:
Hi,

I tried to load an aiger file in binary format with the following commands:

import aiger
aig1 = aiger.load('path_to_file',rule='aig') 

This lead to :
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/aiger/parser.py", line 334, in load
    return parse(''.join(f.readlines()), to_aig=to_aig)
  File "/usr/lib/python3.7/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x91 in position 203: invalid start byte

Are aig files supported? If so, what is the correct way of loading them? Thanks!

allrtaken, 2020-02-04:
I tried 
aig1 = aiger.load('path_to_file')

as well, and it did not work either.

mvcisback, 2020-02-04:
Hi @allrtaken,

Unfortunately I haven't implemented a parser for binary AIGER (`.aig`'s). I have implemented a parser for ascii AIGER (`.aag`) and I don' t think it'd be to hard to get this working.

Do you need programmatic writing and reading of `.aig` or is it enough to use another tool to convert from `.aig` to `.aag`. If so I recommend using the `aigtoaig` tool that ships with the AIGER library:
http://fmv.jku.at/aiger/

That said, I can take a look at implementing this weekend.

allrtaken, 2020-02-04:
Hi @mvcisback ,

Thanks for the quick response! I will use aigtoaig for now, so no rush for implementing the aig parser. But in general I'm worried that the size of the ascii files might blow up for the larger models. Also, since ABC only accepts the binary format, it can get tedious to have to convert between one and the other when switching between tools. So it will be good to have that feature as and when its convenient to you.

One question: is there any documentation for py-aiger apart from that in the Readme file?

Thanks!

mvcisback, 2020-02-04:
@allrtaken How big of circuits are you thinking? 10,000s or 100,000s, or more? Currently py-aiger struggles parsing too big of files since it's currently written recursively. See #77. I have been meaning to re-write some of the parsing/writing code anyway, so this might be a good time to fix some of these issues.

As for `abc`.  It's just a wrapper around the command line, but you might find this useful:

https://github.com/mvcisback/py-aiger-abc

It does the aigtoaig conversion for you.

mvcisback, 2020-02-04:
@allrtaken also regarding documentation. Unfortunately no. I've tried to make the README a pretty good overview, but I've been looking to add a readthedocs.

If there are any confusion about how to use different parts of the library please report back and I'll try to document.

allrtaken, 2020-02-04:
> @allrtaken How big of circuits are you thinking? 10,000s or 100,000s, or more? Currently py-aiger struggles parsing too big of files since it's currently written recursively. See #77. I have been [...]

I'm working on benchmarks from the Hardware Model Checking Competition, and I think some of them run into 100000s. For now, I'm working on the smaller ones which should not be too big in aag format as well. I'll check out py-aiger-abc -- thanks!

allrtaken, 2020-02-04:
> @allrtaken also regarding documentation. Unfortunately no. I've tried to make the README a pretty good overview, but I've been looking to add a readthedocs.
> 
> If there are any confusion about how to use different parts of the library please report back and I'll try to document.

I was running into a few problems with some of the functions -- I'll raise a separate issue for them.

closed 2022-08-10 by mvcisback