mvcisback/py-aiger #126: Multi-output AIG

sventhijssen, 2022-09-26:
Dear,

I had a question regarding the reading of AIGs. Say I have a multi-output AIG, how do I convert this into multiple Boolean expressions (BoolExpr)?

In the examples, I only saw code for single-output AIGs:
```
expr10 = aiger.BoolExpr(circ)
```

I have tried to debug the code and change the following in the code (expr.py):

`__call__` on line 23:
```
exprs = dict()
for output in self.output:
    exprs[output] = self.aig(inputs, lift=lift)[0][output]
return exprs
```

and `output(self)` on line 47:
```
def output(self):
    return self.aig.outputs
```

However, this does not seem to work and I have difficulty with the debugging. Could you help me further please?

Thank you in advance.
Best regards,
Sven

mvcisback, 2022-09-26:
Hi @sventhijssen,

The `BoolExpr` API is designed for single output so that it has Boolean semantics.

If you are trying to doing something like word level semantics you might want to consider the wrapper library: `aiger-bv`: https://github.com/mvcisback/py-aiger-bv

There you can use the `concat` method of `UnsignedBVExpr` to achieve this.

closed 2022-09-26 by mvcisback

sventhijssen, 2022-09-27:
Hi @mvcisback, thank you for your response and explanation!