mvcisback/py-aiger #128: Issue parsing aag file

masinag, 2023-01-12:
Hi, 
the function
https://github.com/mvcisback/py-aiger/blob/1f1d95265c9744c605011fd80e2c139745a78e5e/aiger/aig.py#L283

fails to parse some aag files, such as 
https://raw.githubusercontent.com/arminbiere/dualiza/master/aigs/0001.aag

I think the issue is on line https://github.com/mvcisback/py-aiger/blob/1f1d95265c9744c605011fd80e2c139745a78e5e/aiger/parser.py#L245

since these files do not contain a symbol table ([which should be optional](https://github.com/arminbiere/aiger/blob/5f8ce5ca5fc42a06e5d713309f99195a740ac2b4/FORMAT#L148)) and the parser at that point is still the and_parser.

mvcisback, 2023-01-13:
Thanks! I will try reproducing tonight.

masinag, 2023-01-14:
Hi, I added a failing test showing the issue.

mvcisback, 2023-01-14:
Thank you! Fixed in 241a6000ef654d3449c2b53f8f3f5d1005f69919

closed 2023-01-14 by mvcisback

mvcisback, 2023-01-14:
Released in v6.2.0