mvcisback/py-aiger #133: Speed up topsort in parsing

masinag, 2023-12-11:
Currently the parser uses the toposort_flatten from the toposort library
https://github.com/mvcisback/py-aiger/blob/19b3fe3837c3187dec08a5b58ba7cb36ca107c8d/aiger/parser.py#L367

However, I noticed that this really slows down the parsing of files with many gates, e.g. [this file](https://github.com/yogevshalmon/allsat-circuits/blob/main/benchmarks/islis_benchmarks/arithmetic_or/div/div.aag).

After some investigation, I noticed that the implementation of topsort in the toposort library is very inefficient, as it is O(n^2) instead of O(n+m). 
https://gitlab.com/ericvsmith/toposort/-/blob/master/src/toposort.py?ref_type=heads#L47

I suggest either implementing a simple topsort routine or using the [graphlib library](https://docs.python.org/3/library/graphlib.html). What do you think?

mvcisback, 2023-12-12:
thanks for looking into this. Would you be open to creating a PR?

masinag, 2023-12-12:
I found out that graphlib is supported only from python 3.9, I'm looking for an alternative

masinag, 2023-12-12:
I can implement a simple topsort function instead of relying on an external library, what do you think?
It shouldn't be longer than 10 sloc

closed 2023-12-13 by mvcisback