mvcisback/py-aiger #135: Very slow AIG walking

masinag, 2024-11-01:
Hi, I am using py-aiger to perform some AIG manipulation.
In particular, [I am trying to convert AIG to PySMT formulas](https://github.com/masinag/allsat-cnf/blob/main/benchmark/benchmark/io/aig_adapter.py).
I have found an example of AIG waking in https://github.com/mvcisback/py-aiger/blob/f50171549781e0910cd5aec25dd8582a9d219b84/aiger/writer.py#L41

I have tried to adapt this to my scenario, but I have noticed this gets very slow on some instances, e.g. https://github.com/yogevshalmon/allsat-circuits/blob/b57c2d6cba244460008dc6400beef2604a720c24/benchmarks/random_aig/large_cir_or/bench1/bench1.aag

It seems to me that the bottleneck is somewhere in `aiger.common.dfs` function, likely operations on sets of nodes.
I suspect this can be due to the computation of hash for nodes (generated by `@attr.frozen`), which traverses the whole subgraph each time, for each node. 

I attach code to replicate the issue. 

```python    
import aiger
import funcy as fn

def gates(circ):
    gg = []
    count = 0

    class NodeAlg:
        def __init__(self, lit: int):
            self.lit = lit

        @fn.memoize
        def __and__(self, other):
            nonlocal count
            nonlocal gg
            count += 1
            new = NodeAlg(count << 1)
            right, left = sorted([self.lit, other.lit])
            gg.append((new.lit, left, right))
            return new

        @fn.memoize
        def __invert__(self):
            return NodeAlg(self.lit ^ 1)

    def lift(obj) -> NodeAlg:
        if isinstance(obj, bool):
            return NodeAlg(int(obj))
        elif isinstance(obj, NodeAlg):
            return obj
        raise NotImplementedError

    start = 1
    inputs = {k: NodeAlg(i << 1) for i, k in enumerate(sorted(circ.inputs), start)}
    count += len(inputs)

    omap, _ = circ(inputs=inputs, lift=lift)

    return gg

def main():
    circ = aiger.to_aig("bench1.aag")
    gg = gates(circ)

    print(len(gg))

if __name__ == '__main__':
    main()
```

mvcisback, 2024-11-04:
Hi @masinag ,

Thanks for reaching out. I can take a look sometime in the coming weeks.

I suspect you're right about the hash issue, it's been a bit of a wart for a while and one of the reasons the lazy API was initially developed -- although that won't help here.

I recommend looking at it with a tool like pyspy to get a flame graph is probably going to good to confirm.

https://github.com/benfred/py-spy

If you have a chance to take a look at the py-spy let me know (feel free to attach the output svg).

Supposing it *is* the hashing in `common.dfs` we can look at two solutions:

1. accelerating hashing in general.
2. re-writing `common.dfs` to avoid the hashing.

# Option 1
For option 1, I would have thought this was solved by `cache_hash`. 

https://github.com/mvcisback/py-aiger/blob/f50171549781e0910cd5aec25dd8582a9d219b84/aiger/aig.py#L53

Perhaps we're having a lot of hash collisions and being killed by equality checks? Eitherway it's strange worst case we'll need to manually introduce smarter hashing and caching.

# Option 2
I think this is the easiest to code, but not a very satisfying solution. Essentially would could switch to checking if that *exact* node has already been emitted. This would be done perhaps as follows.

```python
def dfs(circ):
    """Generates nodes via depth first traversal in pre-order."""
    emitted: set()
    stack = list(circ.cones | circ.latch_cones)

    while stack:
        node = stack.pop()

        if id(node) in emitted:
            continue

        remaining = [c for c in node.children if id(c) not in emitted]

        if len(remaining) == 0:
            yield node
            emitted.add(id(node))   # node -> id(node)
            continue

        stack.append(node)  # Add to emit after remaining children.
        stack.extend(remaining)
```

masinag, 2024-11-04:
> @masinag looking at your code again, it may actually be that `NodeAlg` doesn't cache its hashes. Could you try again with that?

I don't think I understand what you mean. I don't see where I am hashing `NodeAlg` objects

masinag, 2024-11-04:
About the proposed options, I can try to profile the execution with py-spy. 

Option 2 seems an easy fix, but if the issue is really the hashing, speeding it up could improve performance in many other contexts. So it could be worth looking deeper into that.

mvcisback, 2024-11-05:
> > @masinag looking at your code again, it may actually be that `NodeAlg` doesn't cache its hashes. Could you try again with that?
> 
> 
> 
> I don't think I understand what you mean. I don't see where I am hashing `NodeAlg` objects

Err, actually ignore what I said.

masinag, 2024-11-22:
Hi, I profiled the above code using py-spy. I stopped it after 3 hours of execution. This is the flame graph.

[image]

It looks like the problem is the equality check between nodes, since most of the time is taken by the `__eq__` function generated by `attrs`.

mvcisback, 2024-11-22:
Thanks @masinag ! Could you share the SVG as well since it's interactive?

But that I suppose this makes sense. It's not the hash that's the problem, it's the equality check that happens after to check it wasn't a hash collision....

I will need to think about how to speed that up. Given that 

https://github.com/mvcisback/py-aiger/blob/f50171549781e0910cd5aec25dd8582a9d219b84/aiger/common.py#L192

seems to be the bottleneck, it make be good to implement option 2 anyway.

Alternatively, I could do a breaking change and make equality the same as an `id` check. It's been a while since I really thought about what implications that would have, but it seems like a reasonable option.

mvcisback, 2024-11-22:
@masinag if I could ask for a favor, if you have the bandwidth, could you regenerate the above flameplot with the change suggested in:

https://github.com/mvcisback/py-aiger/issues/135#issuecomment-2455686829

I suspect we'll see a huge speed improvement, but unfortunately I'm not able to run it myself for a while.

masinag, 2024-11-22:
> Thanks @masinag ! Could you share the SVG as well since it's interactive?

Sure, you can find it at https://github.com/user-attachments/assets/b11cc6ba-1966-4108-b447-e6d3997d51ba

masinag, 2024-11-22:
> @masinag if I could ask for a favor, if you have the bandwidth, could you regenerate the above flameplot with the change suggested in:

I've stopped it after 30 min.

[image]

(interactive version https://github.com/user-attachments/assets/e000cc4a-aaab-4274-9193-3db3fdc4dc17)

It looks like there is another call to __eq__ somewhere else, I am not sure where

mvcisback, 2024-11-22:
Amazing! This does suggest to me that the right solution might be to do a breaking change and make `__eq__` work via `id`. This would have have the same effect (since after hashing, eq is called), but would be done throughout the codebase.

I'll think about it over the weekend to make sure there aren't any gotcha's to applying more widely.

masinag, 2024-11-22:
Great! Btw from [attrs' doc](https://www.attrs.org/en/stable/hashing.html):

> If you want hashing and equality by object identity: use @define(eq=False)

mvcisback, 2024-12-04:
Hi,

sorry for the slow follow up. I tried implementing this and it seemed to break the unit tests (run via `pytest`).

@masinag when you did you change, did you happen to run the `pytest` suite? I'm worried there's a subtle issue I couldn't figure out in the 30min I could dedicate to exploring this.

masinag, 2024-12-04:
I confirm that the tests are failing also for me. I don't have much time to debug it now, but the failing tests are very small, so it should be easy to spot the bug.

masinag, 2024-12-14:
Hi, I have done some debugging and I found out this.
In `tests.test_common.test_tee`, at some point `__call__` is called on a circuit with the following `node_map`:
```python
pmap({
    'b': Input(name='562f849c-ba08-11ef-b60b-6c24083d2fd2'), 
    'a': Input(name='562f849c-ba08-11ef-b60b-6c24083d2fd2'), 
    'c': Input(name='562f849c-ba08-11ef-b60b-6c24083d2fd2')
})
```
So we have three inputs with the same name but different keys in the nodes map, and also different `id(node)`:
`[127306698814800, 127306698805456, 127306698802512]`.

However, since they have the same name, these nodes are considered as if they were the same node by the `dfs`, and only `b` is visited and then stored in `store` by `__call__`.

I am not really sure how this test is built, and what the expected behavior is.

masinag, 2024-12-14:
For instance, can we use the `node.name` as key instead of its `id`?

mvcisback, 2024-12-16:
Thanks @masinag .

I will take a look today.

mvcisback, 2024-12-16:
@masinag I created a quick PR to try out this idea.

It seems to pass all the tests (except for hash stability which is expected given the breaking change.)

Do you mind benchmarking when you have some time? I need to context switch back to my dayjob :)

mvcisback, 2024-12-17:
@masinag had some free time to sitdown and try.

Looks like the issue is resolved. I added a performance test that checks that your example is fast.

closed 2024-12-17 by mvcisback

masinag, 2024-12-17:
Wonderful :tada:

PR #136 (2024-12-17): perf: Switch to using ids for hashing and eq check.
closes: #135
BREAKING_CHANGE: Structually equal circuits will not necessarily be equal now.
commit a7e63cec (2024-12-17): perf: Switch to using ids for hashing and eq check. (#136)

closes: #135
BREAKING_CHANGE: Structually equal circuits will not necessarily be equal now.